PelionIoT/mbed-cloud-client #18: sotp_flash_write_area expects less than 8 byte flash page size

toyowata, 2018-10-04:
This is reported by Renesas during to consider increasing internal flash page size (currently =1 and wanted to increase 32 bytes to speed-up firmware update);

When they set 32 for the flash page size, they got page alignment error at runtime here:

https://github.com/ARMmbed/mbed-cloud-client/blob/master/mbed-client-pal/Source/PAL-Impl/Modules/Storage/Flash/pal_internalFlash.c#L399

This is because the `sotp_flash_write_area()` is called by `write_record()` function here:

https://github.com/ARMmbed/mbed-cloud-client/blob/9b0bc6a2f3f1f5dbb5be1827db83004d531b99c9/factory-configurator-client/mbed-client-esfs/source/sotp/sotp.c#L339

And the flash write size (`(uint32_t *)&header`) is 8-bytes because this is size of the `record_header_t` structure

https://github.com/ARMmbed/mbed-cloud-client/blob/3137b3d7c8da10bae38749a9384e5e5c77a55012/factory-configurator-client/mbed-client-esfs/source/sotp/sotp_int.h#L43-L47

Therefore, when platform flash page size (return value of `flash_get_page_size()`) is bigger than 16, it always generate alignment error.

Possible case platforms:
* Arm MPS2 (256 bytes)
* NXP LPC54xxx (256 bytes)
* Toshiba TMPM46B (16 bytes)

maclobdell, 2018-10-13:
I encountered this problem with LPC546XX.  This is blocking the ability to get the latest cloud client with firmware update capability working on LPCXpresso54628 and FF-LPC546XX.  cc @MarceloSalazar @ashok-rao

toyowata, 2018-10-16:
@jenia81 The `pal_internalFlash.c` is recently updated and the line number is changed.  You can find the alignment check here:

https://github.com/ARMmbed/mbed-cloud-client/blob/3bc44b51ff66c11ccc5ee27b358ea6c1caf01dbc/mbed-client-pal/Source/PAL-Impl/Modules/Storage/Flash/pal_internalFlash.c#L443

I created the test program to reproduce this issue in the  branch here:
https://github.com/toyowata/simple-mbed-cloud-client-template-restricted/tree/page_size_test

Step to reproduce the issue:
```
$ git clobe -b page_size_test https://github.com/toyowata/simple-mbed-cloud-client-template-restricted/
$ cd simple-mbed-cloud-client-template-restricted
$ mbed deploy
# add you dev certifiate
$ mbed compile -m k64f -t gcc_arm -f
```

I got a log below:

```
[BOOT] Mbed Bootloader
[BOOT] ARM: 00000000000000000000
[BOOT] OEM: 00000000000000000000
[BOOT] Layout: 0 70E4
[BOOT] Active firmware integrity check:

[BOOT] [++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++]
[BOOT] SHA256: C3AD3A531CFCB368A58C1BBAF2A95E86DE3469DB5F4CBF459C171A8A07105D3C
[BOOT] Version: 1539664090
[BOOT] Slot 0 is empty
[BOOT] Active firmware up-to-date
[BOOT] Application's start address: 0xA400
[BOOT] Application's jump address: 0xAAB9
[BOOT] Application's stack address: 0x20030000
[BOOT] Forwarding to application...

Starting Simple Pelion Device Management Client example
Connecting to the network...
Connected to the network successfully. IP address: 10.128.4.78
[SMCC] Autoformatting the storage.
[SMCC] Reset storage to an empty state.
[SMCC] Failed to delete storage - 4
[SMCC] Starting developer flow
[SMCC] Failed to load developer credentials - is the storage device active and accessible?
Pelion Client initialization failed (1)
```

When I print `address` and `size` value below just before this [line](https://github.com/ARMmbed/mbed-cloud-client/blob/3bc44b51ff66c11ccc5ee27b358ea6c1caf01dbc/mbed-client-pal/Source/PAL-Impl/Modules/Storage/Flash/pal_internalFlash.c#L443):
```
printf("address = 0x%x, size = %d\n, address, size);
```

I got a log below:

```
[BOOT] Mbed Bootloader

(snip)

Starting Simple Pelion Device Management Client example
Connecting to the network...
Connected to the network successfully. IP address: 10.128.4.51
address = 0x8010, size = 8
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
address = 0x8018, size = 16
```

adamelhakham, 2018-10-18:
Hey,
I created a fix. I tested it on linux with 16, 32 byte pages (sotp emulator). I only have K64f and ODIN here (page sizes 8 and 1 respectively). It would really help me if you could try it on a board with 16/32 byte page..

* Instead of using regular esfs repo, try my fork:
`git@github.com:adamelhakham/mbed-client-esfs.git` on branch `sotp-page-fix`.
Replace mbed-cloud-client/factory-configurator-client/mbed-client-esfs with my fork.
* Also use my branch in mbed-client-pal instead of whatever you're using:
`git@github.com:ARMmbed/mbed-client-pal.git` on branch `sotp-page-fix`.
Replace mbed-cloud-client/mbed-client-pal with my branch.

Please let me know if this works for you :)

toyowata, 2018-10-23:
@adamelhakham Thank you very much for the fix.

I tested this by `mbed-cloud-client-example` with your patch (target : Renesas GR-LYCHEE).
```
mbed-cloud-client-example (#528d3d58933a, tag: 2.0.1.1)
|- drivers/esp32-driver (#e314d33ceaa3)
|- drivers/wifi-x-nucleo-idw01m1 (#a00282000178, tag: v1.2.0)
|- mbed-cloud-client (#62de4b89d235, tag: 2.0.1.1)
`- mbed-os (#c53d51fe9220, tags: latest, mbed-os-5.10.1)
```

Both 16 and 32 byte page_size worked fine.  I tried to set 256 byte which is same value for `NXP  LPC54xxx` and `Arm CM3DS_MPS2`, and got following runtime error.

```
mcc_platform_storage_init() - bd->size() = 2027945984
mcc_platform_storage_init() - BlockDevice init OK.
Application ready. Build at: Oct 23 2018 19:28:10
Mbed OS version 5.10.1
Start simple mbed Cloud Client
fcc_init failed with status 4! - exit
application_init_fcc fcc_init failed with status 4! - exit
Failed initializing FCC
Initialization failed, exiting application!
```

adamelhakham, 2018-10-23:
@toyowata in mbed-client-pal/Source/PAL-Impl/Modules/Storage/Flash/pal_internalFlash.c
could you change the PAL_MAX_PAGE_SIZE define to 256 and tell me if it works please?

adamelhakham, 2018-10-23:
@toyowata I pushed another fix to my branch, basic usage seems to work but I still have some issues.. will hopefully fix them tommorow

toyowata, 2018-10-24:
@adamelhakham 
> could you change the PAL_MAX_PAGE_SIZE define to 256 and tell me if it works please?

I confirmed that it fixed the issue with 256 bytes page size.  Thank you very much.

adamelhakham, 2018-11-06:
@toyowata I Fixed the SOTP and merged it into master of the ESFS repo:
https://github.com/ARMmbed/mbed-client-esfs
Fixed the PAL issue, it is still in a branch sotp-page-fix in PAL repo:
https://github.com/ARMmbed/mbed-client-pal

I made quite a few additional changes in the SOTP code, all my tests have passed but I would really appreciate it if you could retest it with your board before we add it to the mbed-cloud-client release.
Thanks :)

toyowata, 2018-11-09:
@adamelhakham Thank you very much for your update.  I tested PDMC example with your latest patch on Renesas GR-LYCHEE (page_size = 256 this time) and worked fine!

Message log is below:

```
[BOOT] Mbed Bootloader
[BOOT] ARM: 00000000000000000000
[BOOT] OEM: 00000000000000000000
[BOOT] Layout: 0 1800D06C
[BOOT] Active firmware integrity check:
[BOOT] [++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++]
[BOOT] SHA256: 038556F6DC1F23DDBE9D3C3612ACB93E45805DA824B0F914CCE89D51A30FB6E2
[BOOT] Version: 1541754485
[BOOT] Slot 0 is empty
[BOOT] Active firmware up-to-date
[BOOT] Application's start address: 0x18010400
[BOOT] Application's jump address: 0x18010400
[BOOT] Forwarding to application...

mcc_platform_storage_init() - bd->size() = 2027945984
mcc_platform_storage_init() - BlockDevice init OK.
Application ready. Build at: Nov  9 2018 18:08:03
Mbed OS version 5.10.1
Start simple mbed Cloud Client
Using hardcoded Root of Trust, not suitable for production use.
Starting developer flow
Developer credentials already exist, continuing..
mcc_platform_init_connection()
NSAPI_STATUS_CONNECTING
NSAPI_STATUS_GLOBAL_UP
Network initialized, connecting...

Client registered
Endpoint Name: 0166a02441d600000000000100100349
Device Id: 0166a02441d600000000000100100349
```

jenia81, 2018-11-22:
@toyowata The fix is delivered as part of Pelion Device Management Client R2.1.0-RC3:
https://github.com/ARMmbed/mbed-cloud-client-example-sources-internal/tree/release-2.1.0

Closing this issue

teetak01, 2018-11-22:
The fix will be part of the next planned release (2.1.0).

teetak01, 2018-12-21:
Closing this as fixed as 2.1.0 has been released. @toyowata please open a new issue if further problems surface.

closed 2018-12-21 by teetak01