PelionIoT/mbed-cloud-client #31: mbed-cloud-client example run in ARM9 system problem

mjchen1, 2019-02-21:
1. I use "python pal-platform/pal-platform.py fullbuild --target ARM_OpenWRT_mbedtls --toolchain GCC-OPENWRT --external ./../define.txt --name mbedCloudClientExample.elf" can build success.
but when I run ".elf" in my system will occur error.
2. when I run "/out/Release/mbedCloudClientExample.elf" will occur "MBEDTLS_ERR_ECP_INVALID_KEY".
this error come from the function "static int ecp_check_pubkey_sw( const mbedtls_ecp_group *grp, const mbedtls_ecp_point *pt )".
3. but, when I run "/out/Debug/mbedCloudClientExample.elf" will occur "MBEDTLS_ERR_X509_INVALID_FORMAT".
This error come from the function "static int x509_crt_parse_der_core( mbedtls_x509_crt *crt, const unsigned char *buf, size_t buflen )"
4. my system cpu is "arm926ej-s" and I run in Linux_4.4 Kernel.
5. why run Release/.elf and Debug/.elf have different result..??
6. and, how can I solve these error...??(any suggestion..??)

teetak01, 2019-02-22:
@mjchen1 I presume you have fresh developer-certificate from Pelion? If the application starts, can you copy here the actual traces the application gives out?

It also would help to enable debug tracing https://github.com/ARMmbed/mbed-cloud-client-example/blob/d111ac1d954ce068884aee2923059f5f98d9b33e/define.txt#L30 by defining that flag as 1.

mjchen1, 2019-02-25:
Yes, I have fresh developer-certificate from Pelion.

I log Release and Debug message.

The privileged confidential information contained in this email is intended for use only by the addressees as indicated by the original sender of this email. If you are not the addressee indicated in this email or are not responsible for delivery of the email to such a person, please kindly reply to the sender indicating this fact and delete all copies of it from your computer and network server immediately. Your cooperation is highly appreciated. It is advised that any unauthorized use of confidential information of Nuvoton is strictly prohibited; and any information in this email irrelevant to the official business of Nuvoton shall be deemed as neither given nor endorsed by Nuvoton.

mjchen1, 2019-02-25:
[release1.log](https://github.com/ARMmbed/mbed-cloud-client/files/2899340/release1.log)
[debug1.log](https://github.com/ARMmbed/mbed-cloud-client/files/2899345/debug1.log)

yogpan01, 2019-02-25:
Hi @mjchen1,
From debug logs it seems that your device is not able to establish TCP connection to Pelion Cloud server, bootstrap.us-east-1.mbedcloud.com on port 5684.
The Device Management Client is able to resolve the server address (2600:1f18:259e:8903::41) but fails to connect over TCP.
Do you know if there is any firewall or is the access on port 5684 blocked, which may prevent your client to connect to Pelion Cloud.

mjchen1, 2019-02-27:
I am connected to the network through a mobile phone, so there should be no firewall problems.
Also, I can ping to the mbed website.

yogpan01, 2019-02-27:
What about the port 5684? Sometimes certain ports are restricted. mbed website is hosted in different domain than Pelion Cloud.

mjchen1, 2019-03-06:
[image]

I use netcat to check port 5684 is ok.
but still have the same error.

yogpan01, 2019-03-06:
Hi,
Can you verify if you are able to connect example to Cloud through Ubuntu desktop environment?

Regards,
Yogesh

mjchen1, 2019-03-06:
I can connect example to Cloud through Ubuntu desktop environment.
And, the desktop also use the same mobile phone network.

yogpan01, 2019-03-06:
So, you are observing the same network error there as well?

mjchen1, 2019-03-06:
I run example well through Ubuntu desktop environment.
But, run example on ARM9 platform fail.

mjchen1, 2019-03-06:
[image]

I run example well through Ubuntu desktop environment.