PelionIoT/mbed-cloud-client #34: TLV encoding?

Alex-EEE, 2019-02-28:
Release notes say:
"Fixed handling of content type format for PUT requests on resource level. Client only accepts text/plain and opaque content-types."

Does it no longer support TLV?

yogpan01, 2019-02-28:
Hi @Alex-EEE, 
This is for M2mResource single instance for which client always supported plain and opaque data types only.
Earlier, the handling was imcomplete and it didnt quite check the content type of incoming PUT request properly.
Now the check is in place, so anything other than plain text or opaque will return error as NOT_SUPPORTED format.

Regards,
Yogesh

Alex-EEE, 2019-02-28:
OMA spec 8.2.5 says " An Object Instance or Resource is written to by sending either a CoAP PUT request to the corresponding path. The request
includes the value to be written in the corresponding Plain Text, Opaque, TLV or JSON format according to the ContentFormat option which MUST be specified [CoAP]. "

So why is TLV not supported for Single Resource?

yogpan01, 2019-03-01:
Hi @Alex-EEE,
We can add the support for TLV in future releases, for now its a new feature request for us. We will take that into our backlog.

Regards,
Yogesh

loverdeg-ep, 2019-03-01:
I thought TLV was deprecated?

Alex-EEE, 2019-03-01:
@loverdeg-ep Sure that's the direction LwM2M is heading, but mbed doesn't have a CBOR serializer yet ( CBOR is what is replacing TLV ).  I actually enabled TLV for resource level in client-lite by only commenting out one line of code, so I figured it would be just as easy here too.  The TLV serializer can definitely serialize a single resource.

teetak01, 2019-08-16:
I think this discussion can be resolved.

ghseb, 2020-10-16:
Hello,

I think this is a valid request. Modifying 

https://github.com/ARMmbed/mbed-cloud-client/blob/4fc8a9c027623f2f7722ea0b2d475e7772352f6d/mbed-client/source/m2mresourcebase.cpp#L722-L725

to

```
            if(COAP_CONTENT_OMA_OPAQUE_TYPE != coap_content_type &&
               COAP_CONTENT_OMA_PLAIN_TEXT_TYPE != coap_content_type &&
               COAP_CONTENT_OMA_TLV_TYPE != coap_content_type &&
               COAP_CONTENT_OMA_TLV_TYPE_OLD != coap_content_type) {
                msg_code = COAP_MSG_CODE_RESPONSE_UNSUPPORTED_CONTENT_FORMAT;
            } else {
```

does resolve the issue for me.

Regards,
Sebastian

Alex-EEE, 2020-10-16:
Thanks @ghseb !  Did you happen to do a few PUTs and see if the client records the correct value?

ghseb, 2020-10-19:
Having `Each TLV entry starts with a Type byte that indicates if the TLV contains an Object Instance, a Resource, multiple
Resources, or a Resource Instance. Object Instance and Resource with Resource Instance TLVs contains other TLVs in
their value.` in the Lwm2m Specification, i think only the `a Resource` case was tested by writing binary data to a resource. But this worked out well.

Regards,
Sebastian

loverdeg, 2020-11-22:
Status of cbor serializer?

jbuckAI, 2021-05-19:
Any updates for TLV support per spec?

teetak01, 2021-05-19:
Hi, we have fixed this for upcoming 4.9.0 release.

teetak01, 2021-06-07:
I'll close this as fixed. Please reopen if issues persist.

closed 2021-06-07 by teetak01