PelionIoT/mbed-cloud-client #35: Receiving ConnectMemoryConnectFail

GregorioMarquez, 2019-03-12:
I am getting the Error MbedCloudClient::ConnectMemoryConnectFail while running the official Connect Guide.
Target: nucleo_f207zg

I have compiled this program both with mbed CLI (ARM-GCC) and the Online Compiler, both getting the same error.

This is what I am getting:

[BOOT] Mbed Bootloader
[BOOT] ARM: 00000000000000000000
[BOOT] OEM: 00000000000000000000
[BOOT] Layout: 0 80079C8
[BOOT] Active firmware integrity check:
[BOOT] SHA256: 7C60E8509CD265EF51AA3A456B6062955CAF5C6F39E49B10BAF395AE01AAE8F0
[BOOT] Version: 1552350966
[BOOT] Slot 0 is empty
[BOOT] Active firmware up-to-date
[BOOT] Application's start address: 0x8010400
[BOOT] Application's jump address: 0x8039E41
[BOOT] Application's stack address: 0x20020000
[BOOT] Forwarding to application...
Starting Simple Pelion Device Management Client example
You can hold the user button during boot to format the storage and change the device identity.
Connecting to the network using the default network interface...
Connected to the network successfully. IP address: 192.168.2.14
Initializing Pelion Device Management Client...
Initialized Pelion Device Management Client. Registering...
**** current_size: 18113
**** max_size    : 20074
[SMCC] Error occurred : MbedCloudClient::ConnectMemoryConnectFail
[SMCC] Error code : 9
[SMCC] Error details : Client cannot connect anymore Memory allocation failed

I have tried increasing the nanostack-hal.event_loop_thread_stack_size with no success. 
How can I fix this? THANK YOU!

teetak01, 2019-03-12:
Hi @GregorioMarquez, it is possible that the application is not able to access the full 128 KB on that board.

For. ex. see https://github.com/ARMmbed/mbed-os/pull/9588 which is similar board and the .bss has been moved to SRAM2.

cc: @MarceloSalazar

GregorioMarquez, 2019-03-14:
Hello @teetak01, I was looking at the memory mapping at the  STM32L475xx and it does have SRAM1 and SRAM2, but the STM32F207xx only have one memory space SRAM, so it is not possible to move the .bss to another memory location. Do you have any recommendations
on this?

GregorioMarquez, 2019-03-14:
Now I am getting this error:

++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0   : 20016810
R1   : 20005480
R2   : 80000000
R3   : 1A4B0B21
R4   : 20005480
R5   : 00000000
R6   : 00000000
R7   : 00000000
R8   : 08053735
R9   : 08053749
R10  : 00000000
R11  : 00000000
R12  : 20005480
SP   : 20005360
LR   : 08049DB3
PC   : 0804985E
xPSR : 41000000
PSP  : 20005340
MSP  : 2001FFC0
CPUID: 412FC230
HFSR : 40000000
MMFSR: 00000000
BFSR : 00000082
UFSR : 00000000
DFSR : 00000000
AFSR : 00000000
BFAR : 1A4B0B29
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --

++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0x8010AEF
Error Value: 0x804985E
Current Thread: nanostack_event_thread  Id: 0x20005640 Entry: 0x802E79D StackSize: 0x3000 StackMem: 0x20002640 SP: 0x2001FF5C 
For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80FF013D

teetak01, 2019-03-21:
@GregorioMarquez you should report this issue to https://github.com/ARMmbed/simple-mbed-cloud-client or to https://github.com/ARMmbed/pelion-ready-example as those repositories handle issues related to the quick examples.

This repository is for issues with the core Device Management Client library.

Alternatively you can also use https://github.com/ARMmbed/mbed-cloud-client-example, although you will need to provide the board configuration yourself. If you can produce the issue with that reference example, then you can open a new issue to that repository. Tutorial for that application can be found from: https://cloud.mbed.com/docs/current/connecting/mbed-os.html and porting guide in https://cloud.mbed.com/docs/current/porting/porting-a-new-target-for-mbed-os.html.

Note that https://os.mbed.com/guides/connect-device-to-pelion does not use https://github.com/ARMmbed/mbed-cloud-client-example

JanneKiiskila, 2019-05-03:
Out of memory issue might be remedied with Client version 3.1.0 release, as the minimal configuration now uses a significantly smaller amount of RAM.
- https://www.pelion.com/docs/device-management/current/cloud-requirements/requirements-for-production-devices.html

GregorioMarquez, 2019-05-27:
The new version did fixed this issue, thanks!

closed 2019-05-27 by GregorioMarquez