PelionIoT/mbed-cloud-client #38: Project Name

40Grit, 2019-04-16:
In a generic sense the name of this project still makes sense in that it enables Mbed OS and MBL to connect to the cloud.

However this is more specifically an LWM2M client.

If there are extensions for Pelion, then it should probaly be called Pelion client.

There are a lot of vestigial Mbed client Pelion Mbed cloud examples and repositories.

Even someone who is somewhat versed in the history of these projects very often ends up at the deprecated client by accident.

GitHub provides an option to archive projects which might help too.

SMCC could also use a rename. Pelion Client "Helper" or some other better word than "simple".

@maclobdell

loverdeg-ep, 2019-04-24:
@teetak01 @yogpan01

JanneKiiskila, 2019-04-24:
The mbed-cloud-client and mbed-cloud-client-example repos will likely remain with the name they have.
- We have a bunch of continuous integration (CI) jobs etc., which are dependent on the repository names etc., and changing those would cause extra work in the CI, which I don't see as a priority item.
- We also have customers who are pulling the code from that repo and well, renaming a repo would break potentially the customer CIs as well.

So, at least this repo and the example will remain with the name they have. 

We can archive the deprecated repositories.

teetak01, 2019-04-24:
I archived the deprecated repositories related to Client.

loverdeg-ep, 2019-04-24:
Github doesn't re-direct renamed projects/repositories?

loverdeg-ep, 2019-04-24:
Archiving the deprecated client repo's is a great step.

I see #38 as a UX nightmare for new developers, that don't understand the separation between Mbed and Pelion and how LWM2M plays into it all.

loverdeg-ep, 2019-04-24:
https://github.com/ARMmbed/simple-cloud-client
The above maybe remains unarchived. 
Maybe needs @janjongboom's permission?

JanneKiiskila, 2019-04-24:
Yes, @janjongboom needs to handle that.

janjongboom, 2019-04-24:
@jannekiiskila I don’t have admin rights for that repo.

MarceloSalazar, 2019-04-24:
simple-cloud-client is now archived.

40Grit, 2019-04-29:
@JanneKiiskila Github doesn't re-direc the URLs of renamed projects/repositories?

janjongboom, 2019-04-29:
@JanneKiiskila @40Grit It does; as long as the old name is not reclaimed.

teetak01, 2019-05-03:
Hi @40Grit we have no intention of renaming repositories at this time. Closing as all other issues are resolved.

closed 2019-05-03 by teetak01