PelionIoT/mbed-cloud-client #47: Difference between mbed-cloud-client and simple-mbed-cloud-client

wajahat-ublox, 2019-06-14:
What exactly is the difference between the two? It seems both repositories are actively maintained and both are intended to be used for Pelion Device Management.

40Grit, 2019-06-14:
Simple Mbed cloud client depends on Mbed cloud client.

Apparently Mbed cloud client is too complex to use on its own so simple Mbed cloud client was created to get us up and running faster.

I tend to agree that Mbed cloud client is pretty complex. I have used it on its own in the past and it took a significant amount of code review to understand how to get it working the way I wanted. I personally believe this is due to some strange interpretations of the lwm2m enabler specification.

It's a pain but I highly recommend reviewing code for both repositories so that feedback can be provided in order to improve the client and it's "helper" repository SMCC.

loverdeg-ep, 2019-08-16:
Any further updates on the differences between the two?

It would probably be nice for people to know for instance that SMCC defaults to using strings for everything.

JanneKiiskila, 2019-08-21:
Hi,

It is true we have two different example applications/client libraries.

[mbed-cloud-client-example](https://github.com/ARMmbed/mbed-cloud-client-example)
- Our default offering, supports all operating systems we support and as such, has then more complicated structure/more files in the root etc.
- But, on the other hand – allows you to do full development using Linux and it’s superior tooling.
-   - You can use things like gov, valgrind etc. for the application development in Linux. 
-    - Similar tools do not exist for embedded systems (like Mbed OS).
-    - For more complex application logic – I would claim – it makes more sense to develop the board-independent application logic using Linux and use the embedded target only for board specific things and final product validation.

However, it has been recognised this complexity might scare a way some people – thus a more simpler example has been created.

[simple-mbed-cloud-client](https://github.com/ARMmbed/simple-mbed-cloud-client) was sprung out as a way to simplify things a bit.
- It is using the mbed-cloud-client directly, it only adds a few wrapper APIs on top of it to simplify things.
- It for example made the resource creation a bit simpler, thus making the main.cpp file shorter.
- This is however something we have recognized and the simpler resource creation API is in the process of being merged to mbed-cloud-client itself.
- The examples using this also only targeted Mbed OS, so their structure and number of files in the root was thus much smaller.
- This repository will be deprecated, as mbed-cloud-client itself will support the same features/functionality.

We are in the process of creating a new example repository for an Mbed OS only example that will be released in the upcoming month or so. This example repo should then also work as the basis for board / vendor specific extra examples, that can illustrate for example the sensors available on those boards etc.

I hope this clarifies things a bit. Any questions / comments or can we close the ticket?

wajahat-ublox, 2019-08-21:
Thank you for sharing the details. Feel free to close it if others do not have any further question.

JanneKiiskila, 2019-08-21:
Thank you, closing ticket.

closed 2019-08-21 by JanneKiiskila