PelionIoT/mbed-cloud-client #49: Release Benchmarks

loverdeg-ep, 2019-06-27:
Can we get RAM/Flash consumption benchmarks with each release of the client?

teetak01, 2019-06-28:
We can start making those regular as part of at least major/minor releases.

JanneKiiskila, 2019-08-08:
Numbers are published in here (for latest release, no intention of keeping a history - but the docs are versioned, so one can shuffle back) https://www.pelion.com/docs/device-management/current/cloud-requirements/requirements-for-production-devices.html

40Grit, 2019-08-08:
That should be sufficient.

However, it would also be beneficial if there were a repository for the documentation sources.

It would be good to have a place to track issues from the community/partners for the docs.

Additionally the diffs between doc releases are much more apparent when viewed from source in github or other git interface.

Even if the doc source releases came squashed similar to the client releases.

JanneKiiskila, 2019-08-08:
@iriark01  @peknis01  @melwee01 - note the feedback on docs.

JanneKiiskila, 2019-08-08:
@40Grit  - docs issues can be raised also here as issues, no problem in that.

melwee01, 2019-08-08:
Noted.

closed 2019-08-08 by JanneKiiskila

iriark01, 2019-08-08:
The repo is private by request of Product.

loverdeg-ep, 2019-08-08:
@iriark01 Could it at least be opened up to partners in some form?

iriark01, 2019-08-08:
That's a product decision. Janne can decide with them, and I'll happily open it (them - there's more than one repo)