PelionIoT/mbed-cloud-client #56: client-lite public release

Alex-EEE, 2019-12-10:
Any update on when client lite will be publicly available?

JanneKiiskila, 2019-12-11:
We are working on it full speed, hoping to have it out before Christmas but can't guarantee that. If it slips Christmas then it will slip to mid January, 2020. The 1st release will be an Alpha release, so it will not be production quality - we will however accept issues & feedback to guide our development.

ghseb, 2020-02-10:
Is there any news on this topic? A light client definitely makes sense for very constrained devices.

JanneKiiskila, 2020-02-10:
Things rarely go as planned, it seems - but we're in the last finalization steps of the release process. Alpha release will be out this week. Memory targets we have will not be met with this release though, some more squeezing to be done (mainly the MbedTLS side).

JanneKiiskila, 2020-02-12:
Hi @Alex-EEE 

Client Lite Alpha 1.0 release is now officially out - please see [docs](https://www.pelion.com/docs/pelion-client-lite/v1.0/introduction/index.html) and [Client Lite example](https://github.com/ARMmbed/pelion-client-lite-example).

closed 2020-02-12 by JanneKiiskila