PelionIoT/mbed-cloud-client #62: dereferencing pointer to incomplete type 'mbedtls_md_info_t' build error

AshUK, 2020-02-17:
Description of defect

Fails to compile with latest tools:

Build error on **MacOS Catalina 10.15.3**

```
[Error] arm_uc_crypto_mbedtls.c@66,77: dereferencing pointer to incomplete type 'mbedtls_md_info_t' {aka 'const struct mbedtls_md_info_t'}
[ERROR] ./src/libs/mbed-cloud-client/update-client-hub/modules/common/source/arm_uc_crypto_mbedtls.c: In function 'ARM_UC_cryptoHashFinish':
./src/libs/mbed-cloud-client/update-client-hub/modules/common/source/arm_uc_crypto_mbedtls.c:66:77: error: dereferencing pointer to incomplete type 'mbedtls_md_info_t' {aka 'const struct mbedtls_md_info_t'}
   66 |     if (hDigest && output && (output->size_max >= (unsigned)hDigest->md_info->size)) {
      |                                                                             ^~

[mbed] ERROR: "/System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python" returned error.
       Code: 1
       Path: "/Users/ash/dev/mbed_porject"
       Command: "/System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python -u /Users/ash/dev/mbed_porject/mbed-os/tools/make.py -t GCC_ARM -m NUCLEO_L496ZG --source . --build ./BUILD/NUCLEO_L496ZG/GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

Target(s) affected by this defect ?

```
TARGET=NUCLEO_L496ZG
TOOLCHAIN=GCC_ARM
ROOT=.
```

Toolchain(s) (name and version) displaying this defect ?

`arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]`

What version of Mbed-os are you using (tag or sha) ?

`mbed-os #e642a7d8b360 (tags: latest, mbed-os-5.15.1)`

What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

```
mbed-cli --version
1.10.2
```

How is this defect reproduced ?

1. create new mbed project
2. add mbed-cloud-cloud lib
3. add SN_COAP_MAX_BLOCKWISE_PAYLOAD_SIZE
4. add and setup mbedtls_mbed_client_config.h
5. run mbed compile
6. observe failure

teetak01, 2020-02-24:
Hi @AshUK thanks for the report.

Can you confirm are you using one of our example applications https://github.com/ARMmbed/mbed-os-example-pelion or https://github.com/ARMmbed/mbed-cloud-client-example? Or did you write your own application with your custom TLS configuration?

mbed-os-example-pelion is especially suited for quick testing with new target boards.

Are you using the latest version (4.3.0) of this library?

AshUK, 2020-02-24:
@teetak01 Thank you for the response it helped confirming our suspicion that the compile issue was indeed related to a miss TLS configuration. After some tracing we discovered it was related to ARM_UC_FEATURE_CRYPTO_PAL being to set 0 during the build tool chain that was incorrectly pulling in a mbed_lib.json file that was removing the ARM_UC_FEATURE_CRYPTO_PAL definition.

closed 2020-02-24 by AshUK