PelionIoT/mbed-cloud-client #63: Backup & restore FCC on production devices

geoffrey-vl, 2020-02-21:
Hi,

We have discovered a mayor problem with our production devices. They're running mbed-os 5.9.2, and are using a SD card. At one point we're running into problems where the storage is no longer writable due to some other configuration problems that we drag along with us over time.
Long story cut short, the easiest way to solve our problem would be that we could load the FCC'd stuff (that is created on the SD during factory) into RAM, format storage, and then redeploy all.

With a quick test in Windows I've copied the WORKING and BACKUP directory into some other SD card, and next inserted that one into the device, but then `fcc_verify_device_configured_4mbed_cloud()` returns an errorcode.

Is there any way we could accomplish this?

yogpan01, 2020-02-21:
Hi Geoffrey,
We have asked our provisioning team to look into this issue. We will reach out to you soon with some updates.

Regards,
Yogesh

jenia81, 2020-02-23:
Hi @geoffrey-vl.
The flow that you describe here should work. We'll try to recreate it here locally too and let you know if it works. Meanwhile, can you please:
1. Verify that `fcc_verify_device_configured_4mbed_cloud()` function works on the same device before you are switching to the new SD card
2. Open the logs in trace level by setting `"mbed-trace.enable": true` and by adding the following macro `"MBED_TRACE_MAX_LEVEL=TRACE_LEVEL_INFO"` and send us the logs output with the failure 
Thanks

geoffrey-vl, 2020-02-24:
Hi @jenia81.
So I'm running our software in development mode which formats and deploys the dev certificates whenever `fcc_verify_device_configured_4mbed_cloud()` fails. So I've starting with a brand new and fresh (development) IoT device, next I booted my device and connected to the cloud so that we're sure it has a fully working chain of trust. After a 2nd boot the device uses its deployed certificates as expected.

Next I:
1. let the device go into sleep
2. removed the sd card and inserted that in my desktop
3. copied the WORKING and BACKUP folder to my PC
4. formatted the SD card with 32kb cluster size
5. copy the BACKUP and WORKING folder back to the empty SD card
6. insert SD card back in device
7. boot device

After boot the application creates empty configuration files, so I confirm that we can access the SD card.

Next the device fails within `fcc_verify_device_configured_4mbed_cloud()` -> `fcc_check_firmware_update_integrity()` -> `verify_firmware_update_certificate()`.

Here is the trace output:
```
[DEBUG] Initialising FCC
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\factory-configurator-client\source\factory_configurator_client.c:81:fcc_init:===>
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\fcc-output-info-handler\source\fcc_output_info_handler.c:459:fcc_init_output_info_handler:===>
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\fcc-output-info-handler\source\fcc_output_info_handler.c:466:fcc_init_output_info_handler:<===
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\factory-configurator-client\source\factory_configurator_client.c:97:fcc_init:<===
[DEBUG] Verifying device cloud certificates...
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\factory-configurator-client\source\factory_configurator_client.c:177:fcc_verify_device_configured_4mbed_cloud:===>
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\fcc-output-info-handler\source\fcc_output_info_handler.c:471:fcc_clean_output_info_handler:===>
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\fcc-output-info-handler\source\fcc_output_info_handler.c:483:fcc_clean_output_info_handler:<===
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\key-config-manager\source\key_config_manager_internal.c:244:_kcm_item_get_data_size:===> item name = mbed.Timezone len=13
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\key-config-manager\source\key_config_manager.c:37:kcm_init:===>
[INFO][esfs]: esfs_init - enter
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\key-config-manager\source\key_config_manager_internal.c:244:_kcm_item_get_data_size:===> item name = renewal_status len=14
[INFO][esfs]: esfs_open - enter
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\key-config-manager\source\key_config_manager.c:58:kcm_init:<===
[INFO][esfs]: esfs_open - enter
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\fcc-output-info-handler\source\fcc_output_info_handler.c:492:fcc_store_warning_info:===> warning_string is Item not set:
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\fcc-output-info-handler\source\fcc_output_info_handler.c:516:fcc_store_warning_info:<===
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\utils\source\common_utils.c:37:fcc_get_kcm_data:===>
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\key-config-manager\source\key_config_manager_internal.c:244:_kcm_item_get_data_size:===> item name = mbed.UTCOffset len=14
[INFO][esfs]: esfs_open - enter
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\fcc-output-info-handler\source\fcc_output_info_handler.c:492:fcc_store_warning_info:===> warning_string is Item not set:
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\fcc-output-info-handler\source\fcc_output_info_handler.c:516:fcc_store_warning_info:<===
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\key-config-manager\source\key_config_manager_internal.c:293:_kcm_item_get_data:===> item name = mbed.UseBootstrap len = 17, data max size = 4
[INFO][esfs]: esfs_open - enter
[INFO][esfs]: esfs_file_size - enter
[INFO][esfs]: esfs_read - enter
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\key-config-manager\source\key_config_manager_internal.c:329:_kcm_item_get_data:<=== kcm data size = 4
[INFO][esfs]: esfs_close - enter
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\key-config-manager\source\key_config_manager_internal.c:244:_kcm_item_get_data_size:===> item name = mbed.EndpointName len=17
[INFO][esfs]: esfs_open - enter
[INFO][esfs]: esfs_file_size - enter
[INFO][esfs]: esfs_close - enter
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\key-config-manager\source\key_config_manager_internal.c:268:_kcm_item_get_data_size:<=== kcm data size = 32
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\factory-configurator-client\source\fcc_verification.c:946:fcc_check_device_general_info:<===
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\key-config-manager\source\key_config_manager_internal.c:244:_kcm_item_get_data_size:===> item name = mbed.Manufacturer len=17
[INFO][esfs]: esfs_open - enter
[INFO][esfs]: esfs_file_size - enter
[INFO][esfs]: esfs_close - enter
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\key-config-manager\source\key_config_manager_internal.c:268:_kcm_item_get_data_size:<=== kcm data size = 15
[... cut ...]
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\fcc-output-info-handler\source\fcc_output_info_handler.c:552:fcc_store_error_info:===> fcc_status is 13
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\fcc-output-info-handler\source\fcc_output_info_handler.c:567:fcc_store_error_info:<===
[ERR ][fcc ]: .\mbed-cloud-client\factory-configurator-client\factory-configurator-client\source\fcc_verification.c:1068:fcc_check_firmware_update_integrity:<=== Failed to verify integrity CA certificate

```

jenia81, 2020-02-24:
@geoffrey-vl The update certificate you are using for the compilation was created on 25th of June 2019. This why it fails in `fcc_verify_device_configured_4mbed_cloud()`. It expects the update certificate to have expiration date higher than the device time which is 24th of February 2020. I'm not sure I understand why it doesn't fail on `fcc_verify_device_configured_4mbed_cloud()` function upon the second boot. Maybe @alzix, that is responsible for the update client, can try to explain this. In any case, can you try to recompile the image with a newer update certificate and tell me if the problem disappears?

geoffrey-vl, 2020-02-26:
Well Pelion Cloud reports that the Dev/bootstrap certificate expires in 2028.
However when I look in my source folder there is also the `.update-certificates` folder which is a leftover from previous OTA fwupdate tests. The default.der is valid from 27 of March 2019 until 25th of June 2019, so that's that. I always believed that the certificate was ignored because you can perfectly delete that file, build the firmware binary and make the device a development IoT device.

So I've recompiled the software with the `.update-certificates` folder (and some other leftovers) removed, and redid the entire fcc development flow again including formatting the disk so that we're sure no old certificates or other leftovers are there on the sd card.

So when the device has booted for the first time `fcc_verify_device_configured_4mbed_cloud()` fails and the app initiates the `fcc_developer_flow()`. Then, when the new credentials have been created we see following output for `fcc_verify_device_configured_4mbed_cloud()`:

```
[DEBUG] Initialising FCC
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\factory-configurator-client\source\factory_configurator_client.c:81:fcc_init:===>
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\fcc-output-info-handler\source\fcc_output_info_handler.c:459:fcc_init_output_info_handler:===>
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\fcc-output-info-handler\source\fcc_output_info_handler.c:466:fcc_init_output_info_handler:<===
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\factory-configurator-client\source\factory_configurator_client.c:97:fcc_init:<===
[DEBUG] Verifying device cloud certificates...
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\factory-configurator-client\source\factory_configurator_client.c:177:fcc_verify_device_configured_4mbed_cloud:===>
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\fcc-output-info-handler\source\fcc_output_info_handler.c:471:fcc_clean_output_info_handler:===>
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\fcc-output-info-handler\source\fcc_output_info_handler.c:483:fcc_clean_output_info_handler:<===
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\fcc-output-info-handler\source\fcc_output_info_handler.c:492:fcc_store_warning_info:===> warning_string is Item not set:
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\fcc-output-info-handler\source\fcc_output_info_handler.c:516:fcc_store_warning_info:<===
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\key-config-manager\source\key_config_manager_internal.c:244:_kcm_item_get_data_size:===> item name = mbed.Timezone len=13
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\key-config-manager\source\key_config_manager.c:37:kcm_init:===>
[INFO][esfs]: esfs_init - enter
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\key-config-manager\source\key_config_manager_internal.c:244:_kcm_item_get_data_size:===> item name = renewal_status len=14
[INFO][esfs]: esfs_open - enter
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\key-config-manager\source\key_config_manager.c:58:kcm_init:<===
[INFO][esfs]: esfs_open - enter
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\fcc-output-info-handler\source\fcc_output_info_handler.c:492:fcc_store_warning_info:===> warning_string is Item not set:
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\fcc-output-info-handler\source\fcc_output_info_handler.c:516:fcc_store_warning_info:<===
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\utils\source\common_utils.c:37:fcc_get_kcm_data:===>
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\key-config-manager\source\key_config_manager_internal.c:244:_kcm_item_get_data_size:===> item name = mbed.UTCOffset len=14
[INFO][esfs]: esfs_open - enter
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\fcc-output-info-handler\source\fcc_output_info_handler.c:492:fcc_store_warning_info:===> warning_string is Item not set:
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\fcc-output-info-handler\source\fcc_output_info_handler.c:516:fcc_store_warning_info:<===
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\key-config-manager\source\key_config_manager_internal.c:293:_kcm_item_get_data:===> item name = mbed.UseBootstrap len = 17, data max size = 4
[INFO][esfs]: esfs_open - enter
[INFO][esfs]: esfs_file_size - enter
[INFO][esfs]: esfs_read - enter
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\key-config-manager\source\key_config_manager_internal.c:329:_kcm_item_get_data:<=== kcm data size = 4
[INFO][esfs]: esfs_close - enter
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\key-config-manager\source\key_config_manager_internal.c:244:_kcm_item_get_data_size:===> item name = mbed.EndpointName len=17
[INFO][esfs]: esfs_open - enter
[INFO][esfs]: esfs_file_size - enter
[INFO][esfs]: esfs_close - enter
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\key-config-manager\source\key_config_manager_internal.c:268:_kcm_item_get_data_size:<=== kcm data size = 32
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\factory-configurator-client\source\fcc_verification.c:946:fcc_check_device_general_info:<===
[... cut ...]
[INFO][fcc ]: .\mbed-cloud-client\factory-configurator-client\factory-configurator-client\source\factory_configurator_client.c:213:fcc_verify_device_configured_4mbed_cloud:<===
```
It states that the UpdateAuthCert is not set, hence why the expiration date check doesn't fail.

**BUT** after having successfully connected to the cloud I rebooted the device yet again, and now the firmware update certificate expiration check fails again: ` Device time is 1582708870, cert time is 1561450860`. Where did it get that from?

geoffrey-vl, 2020-02-26:
One more important thing I've just noticed: when I do the backup-format-restore procedure before the device has connected to the cloud ( and after a new device id was created through the `fcc_developer_flow()` ), I find no problem. After the BACKUP/WORKING folders are resoted I can successfully connect to the cloud. However, when it is connected to the cloud the update certificate is redeployed again and hence next boot the firmware update certificate expiration check fails again.

So it's really the update certificate that's bothering me at this moment.

alzix, 2020-02-26:
Please note that in developer flow update certificate is hardcoded in `update_default_resources.c`
That file is a byproduct of `manifest-tool init` command, which created together with `.update-certificates` directory.
So in order to recover please perform the next steps:

1. reset the storage on a device
1. re-run `manifest-tool init` command and make sure the `update_default_resources.c` file is placed in application sources and picked up by the build command.

> Note: this time consider specifying longer expiration period.

geoffrey-vl, 2020-02-26:
@alzix , you're right, I forget to renew the `update_default_resources.c`. I did that now using manifest tool, next I've repeated the backup-format-restore procedure.
Its working now, thanks!

closed 2020-02-26 by geoffrey-vl