PelionIoT/mbed-cloud-client #65: Auto Observable Allocation Event Handler

AshUK, 2020-04-30:
When setting a resource as auto observable mbed-client fails to allocate memory for the event handler

```
[DBG ][mClt]: M2MInterfaceFactory::create_resource() - creating /32769/0/26241
[DBG ][mClt]: M2MInterfaceFactory::create_object : Name : 32769
[DBG ][mClt]: Created new /32769
[DBG ][mClt]: M2MObject::object_instance(inst_id 0)
[DBG ][mClt]: M2MObject::create_object_instance - id: 0
[DBG ][mClt]: M2MObject::object_instance(inst_id 0)
[DBG ][mClt]: M2MBase::M2Mbase resource name is EMPTY ===========
[DBG ][mClt]: Created /32769/0
[DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 26241)
[DBG ][mClt]: Created new /32769/0/26241
[INFO][APP ]: bytes allocated currently: 17786
[INFO][APP ]: current number of bytes allocated for the heap: 280672
[INFO][APP ]: free bytes on the heap: 262886
[INFO][APP ]: current number of allocations: 105
[INFO][APP ]: number of failed allocations: 0
[INFO][APP ]: cumulative sum of bytes ever allocated: 1143221
[INFO][APP ]: max bytes allocated at a given time: 17944
[INFO][APP ]: threads: name:main, priority: 24, size: 12288, free: 9384
[INFO][APP ]: threads: name:rtx_idle, priority: 1, size: 1408, free: 868
[INFO][APP ]: threads: name:rtx_timer, priority: 40, size: 1536, free: 1392
[INFO][APP ]: threads: name:stm_queue, priority: 24, size: 4096, free: 2888
assertion "_tasklet_id >= 0" failed: file "./src/libs/mbed-cloud-client/mbed-client/mbed-client-classic/source/m2mtimerpimpl.cpp", line 78, function: M2MTimerPimpl::M2MTimerPimpl(M2MTimerObserver&)
```

Mbed-os: 5.15.2
Cloud Client Version: v4.4.0 #64

AshUK, 2020-04-30:
Typical as soon as I opened this issue I managed to solve it. This was due the the MbedCloudClient not yet being initiated before calling set_auto_observable on the resource and thus the shared event stack not been ready. I'm not sure what changed though between 4.3.0 that would cause this.

yogpan01, 2020-05-01:
Hi AshUK,
This is indeed an internal regression on client related with ordering of initialization of internal components. We have identified the issus and fix will be released in the next Client release.

Regards,
Yogesh

teetak01, 2020-08-28:
This has been fixed in 4.5.0 release.

AshUK, 2020-08-28:
👍

closed 2020-08-28 by AshUK