PelionIoT/mbed-cloud-client #68: Thread safety of M2MResourceBase::set_value

Alex-EEE, 2020-07-25:
Hi,

So out of the box, mbed-cloud-client forks several threads that can manipulate the underlying data of the published Objects and Resources.  

Can you comment on the thread safety of using the various get and set functions inside of M2MResourceBase?  Is the data mutexed within the client such that, if the main thread happens to call "set_value" while some event thread happens to be servicing a Lwm2m Write from the network on the same resource, that you won't have a synchronization issue?

teetak01, 2020-08-28:
Hi @Alex-EEE I think this might be valid issue. The values probably could be better protected in case of simultaneous cloud and application access. Have you encountered some issues or is this based on analysis?

teetak01, 2021-09-14:
There will be some concurrency risks in rare cases, but as the LwM2M is designed to report the latest values, not historical series, I think this is acceptable limitation.

closed 2021-09-14 by teetak01