PelionIoT/mbed-cloud-client #72: Different struct layouts in source files cause hard faults

ghseb, 2020-10-08:
Hello,

`MBED_CLIENT_DISABLE_BOOTSTRAP_FEATURE` is defined as `MBED_CONF_MBED_CLIENT_DISABLE_BOOTSTRAP_FEATURE` in https://github.com/ARMmbed/mbed-cloud-client/blob/f72a23e0dc21de4c82ee53fe947153341419a5b9/mbed-client/mbed-client/m2mconfig.h but not all users of `MBED_CLIENT_DISABLE_BOOTSTRAP_FEATURE` include `m2mconfig.h`. Therefore the struct might have a different layout e.g. for `sn_nsdl.c` and `nsdlaccesshelper.c`. The result are hard faults.

https://github.com/ARMmbed/mbed-cloud-client/blob/caf14b6e3117636ea86a3845929ac5c22f303d3f/mbed-client/mbed-client-c/source/include/sn_grs.h#L65-L79

```
[INFO][mClt]: M2MInterfaceImpl::address_ready() Server Type LWM2M
[INFO][mClt]: M2MNsdlInterface::send_register_message()
[INFO][mClt]: M2MNsdlInterface::observation_to_be_sent() - object NULL, in reconnection mode or not registered
[INFO][COAP]: REGISTER MESSAGE: </3/0/5>,</3/0/16>;obs,</3/0/11>;ct="99";aobs="2",</3/0/11/0
[INFO][COAP]: REGISTER MESSAGE: >;obs,</3/0/4>,</5000>;ct="99",</5000/0>;ct="99",</5000/0/1>
[INFO][COAP]: REGISTER MESSAGE: ;obs,</3201>;ct="99",</3201/0>;ct="99",</3201/0/5850>;obs,</
[INFO][COAP]: REGISTER MESSAGE: 3201/0/5853>;obs,</3200>;ct="99",</3200/0>;ct="99",</3200/0/
[INFO][COAP]: REGISTER MESSAGE: 5501>;obs,</1>;ct="99",</1/0>;ct="99",</1/0/8>,</1/0/7>;obs,
[INFO][COAP]: REGISTER MESSAGE: </1/0/6>;obs,</1/0/1>;obs,</1/0/0>;obs
[INFO][COAP]: sn_nsdl_internal_coap_send() - handle: 0x20003d98
[INFO][COAP]: sn_nsdl_internal_coap_send() - handle->grs: 0x20003e00
[INFO][COAP]: sn_nsdl_internal_coap_send() - offsetof(struct nsdl_s, grs): 0x00000018
[INFO][COAP]: sn_nsdl_internal_coap_send() - sizeof(struct nsdl_s): 0x00000040
[INFO][COAP]: __nsdl_c_send_to_server() - nsdl_handle: 0x20003d98
[INFO][COAP]: __nsdl_c_send_to_server() - nsdl_handle->grs: 0x00000000
[INFO][COAP]: __nsdl_c_send_to_server() - offsetof(struct nsdl_s, grs): 0x00000014
[INFO][COAP]: __nsdl_c_send_to_server() - sizeof(struct nsdl_s): 0x0000003c

++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0   : 00000020
R1   : 30020A04
R2   : 200072D8
R3   : 20042AF0
R4   : 200175F8
R5   : 200072D8
R6   : 00000004
R7   : 20042AF0
R8   : 00000180
R9   : 00000000
R10  : 00000180
R11  : 0806ECD4
R12  : 0800F9ED
SP   : 20042AB0
LR   : 080364AD
PC   : 30020A04
xPSR : 20000000
PSP  : 20042A90
MSP  : 2003FF78
CPUID: 410FC241
HFSR : 40000000
MMFSR: 00000001
BFSR : 00000000
UFSR : 00000000
DFSR : 00000000
AFSR : 00000000
Mode : Thread
Priv : Privileged
Stack: PSP
```

Solution would be to directly use `MBED_CONF_MBED_CLIENT_DISABLE_BOOTSTRAP_FEATURE` in `sn_grs.h` or to include `m2mconfig.h` somehow. Eventually other structures are also affected.

teetak01, 2020-10-14:
Hi @ghseb we will check this. It does look like missing header-include there.

teetak01, 2020-10-14:
Internal tracking https://armiot.atlassian.net/browse/IOTCLT-4474

teetak01, 2020-12-01:
Next release 4.7.0 will resolve this.

closed 2020-12-07 by ghseb