PelionIoT/mbed-cloud-client #73: Retransmission buffer size invalidated

ghseb, 2020-10-22:
Hello,
the following line overwrites `coap_s::sn_coap_resending_queue_bytes` and therefore essentially disables retransmissions. Further it invalidates the eventually configured configuration option `MBED_CONF_MBED_CLIENT_SN_COAP_RESENDING_QUEUE_SIZE_BYTES`.
https://github.com/ARMmbed/mbed-cloud-client/blob/f72a23e0dc21de4c82ee53fe947153341419a5b9/mbed-client/source/m2mnsdlinterface.cpp#L273

What i would propose is to remove the function call, because `sn_coap_resending_queue_msgs` and `sn_coap_resending_queue_bytes` is already initialized by configuration options `MBED_CONF_MBED_CLIENT_SN_COAP_RESENDING_QUEUE_SIZE_MSGS` and `MBED_CONF_MBED_CLIENT_SN_COAP_RESENDING_QUEUE_SIZE_BYTES`:
https://github.com/ARMmbed/mbed-cloud-client/blob/4fc8a9c027623f2f7722ea0b2d475e7772352f6d/mbed-coap/source/sn_coap_protocol.c#L161-L164

Regards,
Sebastian

teetak01, 2021-05-19:
Hi @ghseb we'll check this.

Internal ref: https://armiot.atlassian.net/browse/IOTCLT-4879

teetak01, 2021-06-07:
Will be fixed in next public release (4.10.0).

teetak01, 2021-06-18:
@ghseb we already fixed this in 4.9.1 patch release.

ghseb, 2021-06-18:
@teetak01 Thank you!

closed 2021-06-18 by ghseb