PelionIoT/mbed-cloud-client #74: M2MConnectionHandlerPimpl::eventloop_event_handler is called after object instance is deleted

mstarzyk-mobica, 2020-11-06:
Running mbedos 5.15.4 with mbed cloud client 4.5.0

Application running is succesfully connecting to the cloud. Then, when it disconnects, I observe following logs:
[DBG ][mClt]: M2MNsdlInterface::remove_object() 0x20030e18
[DBG ][mClt]: M2MBase::free_resources()
[DBG ][mClt]: M2MBase::~M2MBase() 0x20030e18
[DBG ][mClt]: M2MNsdlInterface::~M2MNsdlInterface() - OUT
[DBG ][mClt]: ~M2MConnectionHandlerPimpl()
[DBG ][mClt]: M2MConnectionHandlerPimpl::eventloop_event_handler 2
[DBG ][mClt]: M2MConnectionHandlerPimpl::socket_connect_handler - _socket_state = 0
[DBG ][mClt]: **~M2MConnectionHandlerPimpl() - OUT**
[INFO][TRANSMISSION]: NSAPI_STATUS_DISCONNECTED
[DBG ][mClt]: **M2MConnectionHandlerPimpl::eventloop_event_handler 66**
[INFO][mClt]: M2MInterfaceImpl::network_interface_status_change - disconnected

The _[INFO][TRANSMISSION]: NSAPI_STATUS_DISCONNECTED_ is printed from the application code, a status update listener:

```c++
  std::atomic<int> net_up{false};

  auto const status_update_listener = [&net_up](nsapi_event_t status, intptr_t param) {
    if(status == NSAPI_EVENT_CONNECTION_STATUS_CHANGE) {
      switch(param) {
        case NSAPI_STATUS_GLOBAL_UP:
          tr_info("NSAPI_STATUS_GLOBAL_UP");
          net_up = NSAPI_STATUS_GLOBAL_UP;
          break;
        case NSAPI_STATUS_LOCAL_UP:
          tr_info("NSAPI_STATUS_LOCAL_UP");
          break;
        case NSAPI_STATUS_DISCONNECTED:
          tr_info("NSAPI_STATUS_DISCONNECTED");
          net_up = NSAPI_STATUS_DISCONNECTED;
          break;
        case NSAPI_STATUS_CONNECTING:
          tr_info("NSAPI_STATUS_CONNECTING");
          net_up = NSAPI_STATUS_CONNECTING;
          break;
        case NSAPI_STATUS_ERROR_UNSUPPORTED:
        default:
          tr_info("NSAPI_STATUS_ERROR_UNSUPPORTED");
          net_up = NSAPI_STATUS_DISCONNECTED;
          break;
      }
    }
  };

  auto net_iface = m_device->get_network_interface();
  net_iface->add_event_listener(status_update_listener);
```

This method is registering the callback on the interface disconnect event, passing "this" as event argument.
Later PAL generates the event with passed 'this' address.
https://github.com/ARMmbed/mbed-cloud-client/blob/f72a23e0dc21de4c82ee53fe947153341419a5b9/mbed-client/mbed-client-classic/source/m2mconnectionhandlerpimpl.cpp#L665

The logs marked in bold suggest that the eventloop_event_handler is called after M2MConnectionHandlerPimpl instance have been deleted. If that is the case, then this call https://github.com/ARMmbed/mbed-cloud-client/blob/f72a23e0dc21de4c82ee53fe947153341419a5b9/mbed-client/mbed-client-classic/source/m2mconnectionhandlerpimpl.cpp#L64
I haven't observed any crashes related to this yet, however the logs suggest an undefined behavior.

To verify my observation I added a simple address logging to the destructor and the event handler.

```c++
extern "C" void eventloop_event_handler(arm_event_s *event)
{
    tr_debug("M2MConnectionHandlerPimpl::eventloop_event_handler %d", event->event_type);

    if (event->event_type != M2MConnectionHandlerPimpl::ESocketIdle) {
        if(!event->data_ptr) {
            tr_error("M2MConnectionHandlerPimpl::eventloop_event_handler event->data_ptr=NULL !!!!");
            assert(event->data_ptr);
        }
        M2MConnectionHandlerPimpl* instance = (M2MConnectionHandlerPimpl*)event->data_ptr;
        tr_debug("instance address: %p", instance);
        instance->event_handler(event);
    }
}
```
```c++
M2MConnectionHandlerPimpl::~M2MConnectionHandlerPimpl()
{
    tr_debug("~M2MConnectionHandlerPimpl()");
    tr_debug("instance address: %p", this);
#if (PAL_DNS_API_VERSION == 2)
    if ( _handler_async_DNS > 0) {
        pal_cancelAddressInfoAsync(_handler_async_DNS);
    }
#endif

    close_socket();
    delete _security_impl;
    _security_impl = NULL;
    pal_destroy();
    tr_debug("~M2MConnectionHandlerPimpl() - OUT");
}
```
The `this` pointer in the destructor is the same one as in the `instance` inside event handler. Logs:
[DBG ][mClt]: ~M2MConnectionHandlerPimpl()
[DBG ][mClt]: instance address: 0x20029d90
[DBG ][mClt]: ~M2MConnectionHandlerPimpl() - OUT
[DBG ][mClt]: M2MConnectionHandlerPimpl::eventloop_event_handler 2
[DBG ][mClt]: instance address: **0x20029d90**
[DBG ][mClt]: M2MConnectionHandlerPimpl::socket_connect_handler - _socket_state = 0
[INFO][TRANSMISSION]: NSAPI_STATUS_DISCONNECTED
[DBG ][mClt]: M2MConnectionHandlerPimpl::eventloop_event_handler 66
[DBG ][mClt]: instance address: **0x20029d90**
[INFO][mClt]: M2MInterfaceImpl::network_interface_status_change - disconnected

teetak01, 2020-12-01:
Hi @mstarzyk-mobica thanks for the report. We'll check this.

teetak01, 2021-05-19:
This was fixed in 4.8.0 release version.