PelionIoT/mbed-cloud-client #75: mbed-os pal_crypto.h file name conflict with cloud-client file

ladislas, 2020-12-01:
### Description of defect

**Note:** this is a duplicate of https://github.com/ARMmbed/mbed-os/issues/13989 to track progress in mbed-cloud-client as well.

(cc @yannlocatelli)

This is the exact same issue as described in https://github.com/ARMmbed/mbed-os/issues/10022.

The issue was fixed in https://github.com/ARMmbed/mbed-os/pull/10042

But then the changes were reverted and the issue brought back with https://github.com/ARMmbed/mbed-os/pull/13217 and https://github.com/ARMmbed/mbed-os/commit/3e011a9c4ac8388bdd50256f961a564b832126b0

On the cloud client end, the PR mentioned by @adamelhakham (https://github.com/ARMmbed/mbed-os/issues/10022#issuecomment-471923066) is nowhere to be found.

#### Target(s) affected by this defect ?

Custom target with BLE and Pelion

#### Toolchain(s) (name and version) displaying this defect ?

```
arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)
```

#### What version of Mbed-os are you using (tag or sha) ?

6.5.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

```
mbed --version
1.10.4

mbed-tools --version
3.5.1.dev36
```

#### How is this defect reproduced ? 

Setup Pelion mbed-cloud-client using a target with BLE.

teetak01, 2020-12-01:
@ladislas could you provide the actual conflict you are seeing in the issue?

ladislas, 2020-12-01:
We have the error message while compiling:

https://github.com/PelionIoT/mbed-cloud-client/blob/f72a23e0dc21de4c82ee53fe947153341419a5b9/mbed-client-pal/Source/PAL-Impl/Services-API/pal_Crypto.h#L22-L24

teetak01, 2020-12-03:
Internal ticket: https://armiot.atlassian.net/browse/IOTCLT-4603

teetak01, 2021-05-19:
will be fixed in 4.9.0 release.

teetak01, 2021-06-07:
@ladislas issue should be resolved now in https://github.com/PelionIoT/mbed-cloud-client/releases/tag/4.9.0

ladislas, 2021-06-07:
Thanks a lot!

closed 2021-06-10 by teetak01