PelionIoT/mbed-cloud-client #77: Security object not lwm2m specification compliant

ghseb, 2020-12-15:
Hello,

to me it seems that the Security Object as implemented in Mbed-Cloud-Client is not compliant to the Lwm2m Specification. Mbed-Cloud-Client defines the following resources for the Security Object: https://github.com/PelionIoT/mbed-cloud-client/blob/master/mbed-client/mbed-client/m2msecurity.h#L38-L58

The Lwm2m Object Registry on the other hand uses those: http://www.openmobilealliance.org/tech/profiles/LWM2M_Security-v1_0.xml

Consider the difference for resource with id 12 (`Bootstrap-Server Account Timeout` vs `OpenCertificateChain`).

Best regards,
Sebastian

ghseb, 2020-12-15:
The additionally added resources (`OpenCertificateChain`, `CloseCertificateChain`, `ReadDeviceCertificateChain`) conflict with resources that were introduced in Lwm2m 1.1 (http://openmobilealliance.org/release/ObjLwM2M_Security/V1_1-20180612-C/OMA-SUP-XML_LWM2M_Security-V1_1-20180612-C.xml)

teetak01, 2021-01-21:
Hi @ghseb thanks for the report. We can review this implementation as part of future work towards new specifications.

ghseb, 2021-01-21:
Thank you, I think this would simplify usage of cloud-client a lot (especially when it comes to bootstrapping).

Do you mind if this issue is left open as a reminder? If you aware of it I also could close it (I assume an eventual change would be rather long term).

teetak01, 2021-01-21:
You can leave it open as a remainder.

jenia81, 2022-05-02:
PDMC 4.12.0 contains a fix for this.
In this version, we introduced LwM2M compliance feature.
The feature is enabled by LWM2M_COMPLIANT compilation flagץ
When the feature is enabled, the `OpenCertificateChain`, `CloseCertificateChain`, `ReadDeviceCertificateChain` aren't in use.

closed 2022-05-02 by jenia81