PelionIoT/mbed-cloud-client #78: Resource 1/0/7 (Server/Binding) should be writable

ghseb, 2020-12-15:
Hello,

according to http://openmobilealliance.org/release/ObjLwM2M_Server/V1_0-20170208-A/OMA-SUP-XML_LWM2M_Server-V1_0-20170208-A.xml resource 7 (Binding) is readable _and_ writable. Therefore at https://github.com/PelionIoT/mbed-cloud-client/blob/master/mbed-client/source/m2mserver.cpp#L63 `M2MBase::GET_PUT_ALLOWED` should be set instead of `M2MBase::GET_ALLOWED`.

Best regards,
Sebastian

teetak01, 2021-01-21:
Hi @ghseb we only allow changing the Binding mode at build-time, thus it is not possible to change in at runtime. In this case this is conscious decision to allow only GET.

ghseb, 2021-01-21:
I see your point. As some background I think the issue came up during bootstrapping. Bootstrapping did not work, because the bootstrap server could not write this resource.

On the other hand if I interpret the specification correctly ` This Resource defines the transport binding configured for the LwM2M Client. If the LwM2M Client supports the binding specified in this Resource, the LwM2M Client MUST use that transport for the Current Binding Mode. ` correctly. One could assume that even if the 1/0/7 would be writable, the client must only respect it, when it is capable to do so.

ghseb, 2021-04-19:
Resolved in mbed-cloud-client-4.8.0 https://github.com/PelionIoT/mbed-cloud-client/pull/82/files#

closed 2021-04-19 by ghseb