PelionIoT/mbed-cloud-client #79: No bootstrapping when EST feature is disabled

ghseb, 2020-12-16:
Hello,

 successfully finishing bootstrapping seems not to be possible when `MBED_CONF_MBED_CLOUD_CLIENT_DISABLE_CERTIFICATE_ENROLLMENT` is defined (which further defines `MBED_CLIENT_DISABLE_EST_FEATURE`). The reason is that `M2MInterfaceImpl::STATE_BOOTSTRAPPED` can only be reached when `M2MInterfaceImpl::finish_bootstrap()` is called in state `ConnectorClient::State_EST_Success`, which obviously is only possible when the EST feature is enabled. One option might be to call `M2MInterfaceImpl::finish_bootstrap()` directly in `M2MInterfaceImpl::bootstrap_finish()` but there might be complications in the EST-enabled case then.

Best regards,
Sebastian

teetak01, 2021-02-12:
We disable this feature by default for Mbed OS and the bootstrap flow is the default mode. I do not see how this would be an issue.

https://github.com/PelionIoT/mbed-cloud-client/blob/master/mbed_lib.json#L17-L21

without EST (which is default in most cases) the code goes actually via void M2MInterfaceImpl::bootstrap_finish() (instead of finish_bootstrap()).

ghseb, 2021-02-12:
Hello @teetak01 

yes, EST disabled by default.

If I remember correctly, the issue in more details was that in 

https://github.com/PelionIoT/mbed-cloud-client/blob/master/mbed-client/source/m2minterfaceimpl.cpp#L457-L469

the branch that calls `M2MInterfaceImpl::bootstrap_done()` is only executed, when the flag `M2MInterfaceImpl::_bootstrapped` is set to true. Which is not the case. Curiously the `_bootstrapped` flag is set in `M2MInterfaceImpl::bootstrap_done()`, but the call to `M2MInterfaceImpl::bootstrap_done()` depends on `_bootstrapped` being true.

teetak01, 2021-02-12:
Are you doing some inter-operabilitity testing with third-party LwM2M server?

ghseb, 2021-02-12:
Well, I used Leshan as Backend.

teetak01, 2021-02-12:
There can be some differences between Pelion and Leshan. We'll keep this in mind.

teetak01, 2021-09-21:
Should be fixed now with 4.11.0 release.

closed 2021-09-22 by ghseb