PelionIoT/mbed-cloud-client #80: Deletion of Server object should be allowed during bootstrapping

ghseb, 2020-12-16:
Hello,

in http://www.openmobilealliance.org/release/LightweightM2M/V1_0_2-20180209-A/OMA-TS-LightweightM2M-V1_0_2-20180209-A.pdf bootstrap delete is specified as:
```
5.2.7.5 BOOTSTRAP DELETE
The Delete operation targets one or several Object Instances and can be sent multiple times.
Only in Bootstrap Interface, the Delete operation MAY target any Instance or all Instances of any Object including the
Security Object (ID:0), supported by the LwM2M Client. The two exceptions are the LwM2M Bootstrap-Server Account and
the single Instance of the Mandatory Device Object (ID:3) which are not affected by any Delete operation.
When the Delete operation is used without any parameter (i.e. without Object ID parameter), all Instances of all Objects in
the LwM2M Client MUST be removed (except for the two cases mentioned above); this functionality could be used for
initialization purpose before LwM2M Bootstrap-Server sends Write operation(s) to the LwM2M Client.
```

with `LwM2M Bootstrap-Server Account` beieng specified as:
```
3.2 Definitions
LwM2M Bootstrap-Server Account LwM2M Security Object Instance with Bootstrap-Server Resource true
LwM2M Server Account LwM2M Security Object Instance with Bootstrap-Server Resource false and associated LwM2M Server
Object Instance
```

In contrast to the specification bootstrap delete on the e.g. object `/1` (`Server`) is effectively prohibited by https://github.com/PelionIoT/mbed-cloud-client/blob/master/mbed-client/source/m2mnsdlinterface.cpp#L2495-L2508

Best regards,
Sebastian

teetak01, 2021-05-19:
Fix for this is planned in future version.

teetak01, 2021-09-21:
Should be fixed now with 4.11.0 release.

closed 2021-09-22 by ghseb