PelionIoT/mbed-cloud-client #87: Abort unregistration process on failure

msiglreith, 2021-07-01:
Hello,

Platform: mbed-client of mbed-cloud-client 4.6 with mbed-os 5.15

We noticed in the mbed-client that in case there is a communication error during `M2MInterfaceImpl::unregister_object` (e.g simulated failure on the server) the client will try to reconnect, which might fail again, etc.

It would be nice to have a proper way to abort the process in this scenario (at least I couldn't find a way right now) as this might lead to increased energy consumption while it may be acceptable to not cleanly deregister from the server after every connection.

Current behavior:
```
[INFO][mClt]: M2MNsdlInterface::send_unregister_message
....
[INFO][mClt]: M2MNsdlInterface::handle_unregister_response - unregistered
[ERR ][mClt]: M2MNsdlInterface::handle_unregister_response - unregistration error 4
[ERR ][mClt]: M2MInterfaceImpl::registration_error code [13]
[ERR ][mClt]: M2MInterfaceImpl::socket_error: (-8), retry (1), reconnecting (0), reconnection_state (2)
[INFO][mClt]: M2MInterfaceImpl::create_random_initial_reconnection_time() initial random time 21
[INFO][mClt]: M2MInterfaceImpl::socket_error - reconnecting in 21(s)
...
```

So far I tried to workaround it by stopping the reconnect attempt during `M2MNsdlInterface::handle_unregister_response`
in case we receive a coap header with status `COAP_STATUS_BUILDER_MESSAGE_SENDING_FAILED`. This allows us to early out of the process.

Is there a better way to handle this or should the client try to reconnect at all during unregistration?

Thanks!

teetak01, 2021-07-02:
Hi @msiglreith we will check the recovery logic in this case. I agree that in case of deregistration, if the request does not go through out nicely, we could just abort and leave it up to the server to clean-up after the device lifetime eventually expires.

teetak01, 2021-07-02:
Internal ref: https://armiot.atlassian.net/browse/IOTCLT-4971

teetak01, 2021-09-21:
Should be fixed now with 4.11.0 release.

msiglreith, 2021-10-04:
thanks again!

closed 2021-10-04 by msiglreith