PelionIoT/mbed-cloud-client #90: Errors when disabling bootstrapping

msiglreith, 2021-08-09:
Disabling the bootstrap features causes some issues with version 4.10.
Below a list of issues I faced so far, but might not be exhaustive:

### Compile-time:
- `ConnectorClient` uses bootstrap related functions which are excluded when disabling the feature
- `ConnectorClient` doesn't derive from `M2MTimerObserver` anymore but relays on observation for some timers.

### Runtime:
- Bootstrap related states in `M2mIntefaceImpl` are excluded, but the states are not excluded as well in the generated state machines, which leads to incorrect state transitions
https://github.com/PelionIoT/mbed-cloud-client/blob/master/mbed-client/source/include/m2minterfaceimpl.h#L503
https://github.com/PelionIoT/mbed-cloud-client/blob/master/mbed-client/source/m2minterfaceimpl.cpp#L160

teetak01, 2021-08-17:
Thanks for the report @msiglreith 

We'll take a look.

Internal ref: https://armiot.atlassian.net/browse/IOTCLT-5013

mirelachirica, 2021-08-23:
>  Bootstrap related states in M2mIntefaceImpl are excluded, but the states are not excluded as well in the generated state machines, which leads to incorrect state transitions

> https://github.com/PelionIoT/mbed-cloud-client/blob/master/mbed-client/source/m2minterfaceimpl.cpp#L160

This whole function is flagged at line 144:
https://github.com/PelionIoT/mbed-cloud-client/blob/d7edc529ed3722c811ff401440ef58ea980bf543/mbed-client/source/m2minterfaceimpl.cpp#L144

I checked the code and all calls for BOOTSTRAP states(STATE_BOOTSTRAP, STATE_BOOTSTRAP_ADDRESS_RESOLVED, STATE_BOOTSTRAP_RESOURCE_CREATED, STATE_BOOTSTRAP_WAIT, STATE_BOOTSTRAP_ERROR_WAIT, STATE_BOOTSTRAPPED) are flagged.

Also state_function() is flagging the bootstrap states:

https://github.com/PelionIoT/mbed-cloud-client/blob/d7edc529ed3722c811ff401440ef58ea980bf543/mbed-client/source/m2minterfaceimpl.cpp#L1375

An error trace could be added to default case when the required state didn't match any of switch cases.

msiglreith, 2021-08-23:
@mirelachirica Thanks for taking a look!

> This whole function is flagged at line 144:

Right, sorry I just picked the first occurence where such state transition arrays are generated. It affects the others as well.
My understanding is that it's related to the way these arrays are generated by the `#defines` defined here:
https://github.com/PelionIoT/mbed-cloud-client/blob/master/mbed-client/source/include/m2minterfaceimpl.h#L650-L658

There is an implicit ordering defined by the state machine definitions (e.g https://github.com/PelionIoT/mbed-cloud-client/blob/master/mbed-client/source/m2minterfaceimpl.cpp#L161-L179)
and the ordering in the `E_States` enum, but this is not enforced at compile-time. 

Example: When defining `MBED_CLIENT_DISABLE_BOOTSTRAP_FEATURE `, `STATE_REGISTER` would be assigned to `1`  by the enum but the corresponding function would be at index `7` in the state machine array (https://github.com/PelionIoT/mbed-cloud-client/blob/master/mbed-client/source/m2minterfaceimpl.cpp#L242). When looking up the transition in the state machine it would pick the transition for `STATE_BOOTSTRAP` (https://github.com/PelionIoT/mbed-cloud-client/blob/master/mbed-client/source/m2minterfaceimpl.cpp#L236)

teetak01, 2021-09-21:
Should be fixed now with 4.11.0 release.

closed 2021-10-04 by msiglreith