PelionIoT/mbed-cloud-client #93: Avoid unnecessary includes

ghseb, 2021-08-25:
Hello,

when `MBED_CLOUD_CLIENT_SUPPORT_UPDATE` is not defined, `UpdateClientResources.h` and `update_client_public.h` should probably not be included in `ServiceClient.cpp` to avoid unnecessary dependencies towards update client hub.

https://github.com/PelionIoT/mbed-cloud-client/blob/d7edc529ed3722c811ff401440ef58ea980bf543/source/ServiceClient.cpp#L28-L31

Proposal:
```
#if !defined(MBED_CLOUD_CLIENT_FOTA_ENABLE)
#if defined(MBED_CLOUD_CLIENT_SUPPORT_UPDATE)
#include "include/UpdateClientResources.h"
#include "update-client-hub/update_client_public.h"
#endif
#else
```

teetak01, 2021-08-25:
Thanks for the feedback @ghseb. Looks sensible.

teetak01, 2021-08-25:
https://armiot.atlassian.net/browse/IOTCLT-5026

teetak01, 2021-09-21:
Should be fixed now with 4.11.0 release.

closed 2021-09-22 by ghseb